GeekFunkLabs/fluidpatcher #10: Delay after not using for a while, not sure if fluidpatcher or fluidsynth

chestercodes, 2021-05-22:
Hi, thanks for your work, it's an awesome project.

I'm trying to get the install working for a raspberry pi and have run the install script from https://geekfunklabs.com/squishbox as per the instructions on a fresh raspberry pi os install.
I plug a USB MIDI controller in and turn it on, it initialises OK, I see the 5 lights flicker and then it starts to make sound. Which is awesome :) 

Having a bit of trouble with notes that are played after a short break. If I don't touch the MIDI keyboard for more than ~15 seconds it will be slow to start making sounds again. It typically makes sounds after a couple seconds of hitting notes, but I've had it not play any single notes and only start making sounds when I make chords.

I've reproduced this behaviour with 2 MIDI keyboards (M-Audio ES61 and Akai LPK25) and have included links to videos of what I mean. It seems a bit less sluggish on the Akai keyboard, although it still shows the behaviour.

[MAudio 61ES](https://www.youtube.com/watch?v=OiDVSA0iuhk)
[Akai LPK25](https://www.youtube.com/watch?v=CGXxmt58VLw)
[Akai LPK25 waiting a minute and then playing again](https://youtu.be/bkj5avGRdwo?t=70)

I'm hoping to use this as a sound module at a gig, and will be switching between playing keyboard and guitar, so i'm hoping there's a bit of config somewhere that can fix this.

I've had a look at the fluidpatcher code and couldn't see anything obviously that would introduce this delay, but I haven't read much python for a while so might have missed something.

Are there any more debugging/troubleshooting things I could do to figure out whether the delay can be removed?

chestercodes, 2021-05-22:
During setup, I selected the options to run it headless, build the latest fluidsynth, download the fonts and update all of the software on the pi.
The videos above are it running headless with the aux jack running into a portable speaker and a single midi keyboard at a time

albedozero, 2021-05-23:
Hmm. It could be a power supply issue - if the Pi isn't getting enough juice it could be rebooting or not supplying steady power to your MIDI controller. What type of Pi are you using, and what is the output current on your adapter? I've gotten weird behavior like this when I tried to use some random phone charger to power the Pi and discovered it was only 1700 mA or something like that. It could be something to do with your speaker as well, though that seems less likely - have you tried it with headphones or a different speaker?

chestercodes, 2021-05-23:
I tried headphones and there's no issue, so i'm guessing that the speaker turns goes to sleep to preserve battery. 
Not a great way to chase my tail for hours, but oh well.

Thanks for your suggestions and work put into the project.

closed 2021-05-23 by chestercodes