GeekFunkLabs/fluidpatcher #13: No sound when trying to use the Novation Launchpad Pro or Circuit Tracks

karhide-xx, 2021-06-17:
Hi,

I thought it would be easier to use GitHub because I have an account.  I have not had time to look at the Raspberry Pi since writing this because of work and having to fix/move a website for a friend that has taken lots of my spare time. 

The one thing I have done is order a Pi Sound sound card by Blokas that I am waiting to arrive which I am hoping will be better than the on board sound and the cheap USB interface I bought.  

Let me know if you have an idea of what  I can test because I have not really got my head round how audio and midi works under Linux because I come from a Windows background. 

Thanks for your help! 

Tim 

Here's where we are from the youtube comments: 

**Karhide**
Just trying the setup now but I don't get any sound when I plug in the midi controller (Launchpad Pro Mk3).  How do I debug because I am very new to audio on a Raspberry Pi.

**The White Tornado**
3 weeks ago
Are you certain you're sending note messages from the Launchpad? The default initial MIDI Mapping on the Mk3 only sends CC (continuous controller) messages - Custom 3 and Custom 4 look like they send some notes. To do some debugging, you can log in to your Pi via ssh (see https://www.raspberrypi.org/documentation/remote-access/ssh/ for how) and type:

sudo systemctl stop squishbox.service
python3 headlesspi.py

This will give you some feedback on when the script changes patches, banks, etc. You can also type `aseqdump -p 20:0` to see what MIDI messages your controller is sending (if your controller is not on port 20:0 type `aconnect -i` to see what it is).
Feel free to reply or post in the forum at https://geekfunklabs.com/forums/forum/questions-issues/ if you have more questions!

**Karhide**
3 weeks ago
 @The White Tornado  Just had a look at the forum and don't see a way to register to ask a question.  I have run the scripts and I can see the launchpad as device 24 and there are note on and off messages.  I have tried changing the channel for the midi notes but nothing happens.  What's the next step to test?

**Karhide**
3 weeks ago
Here's the output  
24:0   Note on                 0, note 51, velocity 25
 24:0   Clock
 24:0   Clock
 24:0   Channel aftertouch      0, value 120
 24:0   Channel aftertouch      0, value 126
 24:0   Clock
 24:0   Channel aftertouch      0, value 127
 24:0   Clock
 24:0   Note off                0, note 53, velocity 0

**Karhide**
3 weeks ago
Here's the midi details:
client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
client 14: 'Midi Through' [type=kernel]
    0 'Midi Through Port-0'
client 24: 'Launchpad Pro MK3' [type=kernel,card=2]
    0 'Launchpad Pro MK3 MIDI 1'
    1 'Launchpad Pro MK3 MIDI 2'
    2 'Launchpad Pro MK3 MIDI 3'

**The White Tornado**
3 weeks ago
Try rebooting the Pi (so the script is running) and log in via ssh again and type `aconnect -iol` - this will let you see if the Launchpad is connecting to FluidSynth. Also, see if your audio output is working by typing `speaker-test`.

**Karhide**
3 weeks ago
 @The White Tornado  It looks like it is connected because the launchpad is 24 (see below) and if I run the speaker test I hear white noise. Is it because the launchpad shows up as multiple items?

client 0: 'System' [type=kernel]
    0 'Timer           '
        Connecting To: 128:0
    1 'Announce        '
        Connecting To: 128:0
client 14: 'Midi Through' [type=kernel]
    0 'Midi Through Port-0'
        Connecting To: 128:0
client 24: 'Launchpad Pro MK3' [type=kernel,card=2]
    0 'Launchpad Pro MK3 MIDI 1'
        Connecting To: 128:0
    1 'Launchpad Pro MK3 MIDI 2'
        Connecting To: 128:0
    2 'Launchpad Pro MK3 MIDI 3'
        Connecting To: 128:0
client 128: 'FLUID Synth (391)' [type=user,pid=391]
    0 'Synth input port (391:0)'
        Connected From: 0:1, 14:0, 24:0, 24:1, 24:2, 0:0

**The White Tornado**
2 weeks ago
Thanks - good to know sound is working and the Launchpad is connecting. It's a bit surprising that the Launchpad presents 3 MIDI devices to read from, but it shouldn't matter - you can connect as many MIDI controllers as you want to FluidSynth and it will read from all of them. I think this must be something to do with the messages the Launchpad is sending. I do notice that the one note on message I can see has a very low volume, so it could be working but the sounds are just quiet. It might also be sending some conflicting messages (like note offs or CC7=0) on the other ports - try `aseqdump -p 24:1` and `aseqdump -p 24:2`. If you have another MIDI controller lying around, try connecting that and see if you get sounds, so you can at least narrow it down to a problem with the Launchpad.

**Karhide**
2 weeks ago
 @The White Tornado  Just had chance to hook it all up again and I can see note on messages with a velocity of 124 but still no sound. I don't have another midi interface with me other than the Novation Circuit Tracks that seems to show no difference.  I do have a midi interface that has 2 ins and  2 outs and I have plugged the launchpad directly in to that and it is just the same. Any ideas or is it time to give up?

**Karhide**
2 weeks ago
My idea would be to use the Fluid synth on headless pi as a sound source for the Circuit Tracks sequencer and I think it'd make a really cool and portable system.

**Karhide**
2 weeks ago
Just done some more digging and reinstalled using the script and then ran fluidsynth and got the following: 

FluidSynth runtime version 2.2.1
Copyright (C) 2000-2021 Peter Hanappe and others.
Distributed under the LGPL license.
SoundFont(R) is a registered trademark of Creative Technology Ltd.

fluidsynth: error: fluid_sfloader_load(): Failed to open '/usr/local/share/soundfonts/default.sf2': File does not exist.
fluidsynth: error: Unable to open file '/usr/local/share/soundfonts/default.sf2'
fluidsynth: error: Failed to load SoundFont "/usr/local/share/soundfonts/default.sf2"

So looks like it could be an issue loading the soundfont.   I will dig a bit more.

**The White Tornado**
2 weeks ago
Hi @karhide - you can ignore the 'unable to load default.sf2' message - the FluidSynth team added an option to load a default soundfont on startup, but our script loads its own soundfonts as specified in SquishBox/banks/bank1.yaml. Not sure why you're still not getting sound - maybe we could hop into discord so we can back-and-forth a bit more easily. I'm on a trip at the moment but will be available in a week.

**Karhide**
1 week ago
 @The White Tornado  That would be amazing if you have time. Just let me know when you are available.

albedozero, 2021-06-29:
Hello - sorry for the delay 😊. Let me get a little info from you and maybe I can help troubleshoot a bit more. First, log in to your Pi and shut down the service that automatically runs the script (it will restart again when you reboot):
```
sudo systemctl stop squishbox.service
````
Now enter `aplay -l` and show me the output - that will help me figure out what your sound card situation is.

Next, run the script from the command line by entering
```
python3 headlesspi.py
```
That should provide a bit more feedback about what's going on with there being no sound.

Just a heads-up - it will probably be easiest to get things working using the Pi's headphone jack as a sound source. If you're using a USB sound card it should work as long as you choose the `USB sound card` option when choosing the audio output in the headless pi setup script, but as every USB sound card is a bit different I can't guarantee. The Blokas card looks cool, but it also looks like they have their own software and drivers to install. I think if you follow their instructions to set up their software, _then_ run the headless pi setup script and choose `No change` for the audio output option it will probably work, but I'd have to do some more digging into their software to get a better idea. Looks like PiSound has a pretty good forum that might help.

karhide-xx, 2021-07-01:
Don't worry about the delay I've been flat out with work and will not be able to look again until next week at the earliest. 

Thanks,

Tim

Ashguitargolf, 2021-07-07:
I have a problem with the pi crashing when I change patches. Today it at least went to 3 other patches before crashing. I'm running the latest Ubuntu desktop on pi 3b+.
I'm a total novice to working with a raspberry pi, only got it last Saturday but am excited to learn python.  @The White Tornado  Ok so I booted up, the piano soundfont voiced when I played the midi keyboard, I entered the command you provided in terminal. The following messages appeared and the sound stopped. "Failed to stop python3.service: Unit python3.service not loaded.
"Failed to stop squishbox.py.service: Unit squishbox.py.service not loaded. Usually when it's headless the pi just crashes as soon as I load a new track. I tried last night on my NI KKm32, and nektar lx49+. The Nektar I pressed every button to change patch with no luck. The M32, one of the encoders changes cc settings, and this is when the pi crashes. Thanks for taking the time to help me get this sorted.