GeekFunkLabs/fluidpatcher #14: I have a problem with the pi crashing when I change patches. Today it at least went to 3 other patches before crashing. I'm running the latest Ubuntu desktop on pi 3b+.

albedozero, 2021-07-07:
I have a problem with the pi crashing when I change patches. Today it at least went to 3 other patches before crashing. I'm running the latest Ubuntu desktop on pi 3b+.
I'm a total novice to working with a raspberry pi, only got it last Saturday but am excited to learn python.  @The White Tornado  Ok so I booted up, the piano soundfont voiced when I played the midi keyboard, I entered the command you provided in terminal. The following messages appeared and the sound stopped. "Failed to stop python3.service: Unit python3.service not loaded.
"Failed to stop squishbox.py.service: Unit squishbox.py.service not loaded. Usually when it's headless the pi just crashes as soon as I load a new track. I tried last night on my NI KKm32, and nektar lx49+. The Nektar I pressed every button to change patch with no luck. The M32, one of the encoders changes cc settings, and this is when the pi crashes. Thanks for taking the time to help me get this sorted.

_Originally posted by @Ashguitargolf in https://github.com/albedozero/fluidpatcher/issues/13#issuecomment-875587342_

albedozero, 2021-07-07:
Hi @Ashguitargolf - I moved this to its own issue to keep things separate.

From the error message it looks like maybe my typed commands didn't come across well in a YouTube comment (and I made a typo too). The commands should be on separate lines, like this:
```
sudo systemctl stop squishbox.service
python3 headlesspi.py
```
The first line stops the script from running in the background, which is needed for running headless. The second starts the script in your terminal window so you can see its error output. Paste what you get from doing that in a reply.

albedozero, 2021-07-07:
Actually, now that I think of it this could be the same problem as what we eventually figured out in #6 where the Pi was rebooting randomly:

> I've had this exact same problem, and it turned out to be my power supply! HDMI draws a lot of power, more so on the newer Pis, and the 2-amp power supply I had just couldn't quite make it if I had a monitor connected.

Ashguitargolf, 2021-07-07:
Fluidsynth: connection of midi through port-0 succeeded
Fluidsynth: connection of Komplete Kontrol M32 midi 1 succeeded
Connection is already subscribed
Fluidsynth warning: Sample ' sinewave' : Rom sample ignored
Loading bank 'bank1.yaml' .. done!
Selected patch 1/26 piano

Ashguitargolf, 2021-07-07:
> Actually, now that I think of it this could be the same problem as what we eventually figured out in #6 where the Pi was rebooting randomly:

Ashguitargolf, 2021-07-07:
The lag when headless is almost not noticeable, with HDMI in, unusable. I did intend for the pi to be a midi host or just plug and play music box. I'm gonna get the pi 4 8gb for a desktop or whatever other project I get inspired to do 
I wonder if you could recommend a good python learning book. I will do some online classes in it, but I'm old skool and having a book as reference is best for me.

albedozero, 2021-07-07:
Sounds good - power supply issue was just a guess.
```
Fluidsynth: connection of midi through port-0 succeeded
Fluidsynth: connection of Komplete Kontrol M32 midi 1 succeeded
Connection is already subscribed
Fluidsynth warning: Sample ' sinewave' : Rom sample ignored
Loading bank 'bank1.yaml' .. done!
Selected patch 1/26 piano
```
So what sort of output do you get from this point if you change patches until you get a crash?

> The lag when headless is almost not noticeable, with HDMI in, unusable.

That's expected. There are many layers of software involved in getting audio into the HDMI stream - the headphone jack is much more low-level. My linux laptop can run the code just fine with no lag on desktop, but the Pi 3+ just doesn't have enough power - a Pi 4 would probably be better but I don't have one yet myself 😉

> I wonder if you could recommend a good python learning book. I will do some online classes in it, but I'm old skool and having a book as reference is best for me.

I'm not formally trained, and it's been an age since I've cracked a programming book, but I always felt the "in a Nutshell" series were pretty accessible, so "Python in a Nutshell" would probably be good. I'm 99% self-taught and do all my learning from web pages - geeks4geeks.org is my latest favorite.

Ashguitargolf, 2021-07-07:
So the piano patch is the first sound I get, that sounds great, and It functions fine until I change patch. I'm just using the headphone jack to a powered small speaker when headless. I'm a bit flummoxed and my programming skills are non existent thus far. I did a programming class in school 30 years ago, so I'm hoping something clicks again when I start learning python, thanks for the tips on how to get started. So I'm thinking what to do next about this patch change problem?

albedozero, 2021-07-07:
Are you getting any more text output from the program when you change patches, or does the Pi just reboot?

Ashguitargolf, 2021-07-07:
When I change patch with terminal open it will show me the different patches, as soon as I play a chord, it takes 2 seconds for it to say shutting down, it doesn't reboot. So piano is safe, the others are dangerous so the pi thinks ha ha

Ashguitargolf, 2021-07-07:
I might give patchbox os a go, see if it works better on that. Have you any experience or knowledge of this os?
It is an os geared towards audio projects.

albedozero, 2021-07-07:
Ah - it sounds like your keyboard might be one of those rare ones that doesn't send a release message when you let go of a button, so the script thinks you're still holding it down, which triggers a shutdown. To see if this is happening, type `aconnect -o` to see a list of your MIDI controllers. You should see something like:
```
client 14: 'Midi Through' [type=kernel]
    0 'Midi Through Port-0'
client 20: 'AKM322' [type=kernel,card=1]
    0 'AKM322 MIDI 1   '
```
Find the client number of your keyboard (mine is 20) and enter `aseqdump -p 20`, then press your patch change button a few times. You should see
```
Waiting for data. Press Ctrl+C to end.
Source  Event                  Ch  Data
 20:0   Control change          0, controller 21, value 127
 20:0   Control change          0, controller 21, value 0
 20:0   Control change          0, controller 21, value 127
 20:0   Control change          0, controller 21, value 0
 20:0   Control change          0, controller 21, value 127
 20:0   Control change          0, controller 21, value 0
```
If you don't see the messages with `value 0` then we've found the problem. You can disable the shutdown feature by commenting out or deleting [this line](https://github.com/albedozero/fluidpatcher/blob/576c85bdd3149077a22640798609824fcfd0fc07/headlesspi.py#L155) in _headlesspi.py_. You won't be able to shut down the Pi when it's headless - you'll just have to unplug it, which has a slight possibility of corrupting the SD card (it's never happened to me though). I might be able to figure out a better fix in time but this would work for now if this turns out to be the issue.

> I might give patchbox os a go, see if it works better on that. Have you any experience or knowledge of this os?
It is an os geared towards audio projects.

I've heard of patchbox but haven't used it. Seems like it's mainly targeted to running apps like Pure Data - which is kind of a synthesizer programming language. I've played around with Pure Data - very fun and interesting but lots to learn. Not sure how fluidpatcher would work in that environment but might be worth looking into in the future...

Ashguitargolf, 2021-07-07:
Ok I got the channel 0 messages, controller 21, as I went up patches it was going up 23,24,25 etc in the value column, then it shutdown doh. How do I access just the raw script to take out the shutdown line, how do I save and exit, I know very few commands? Thanks so much for helping me. I want this to work so my girlfriend can learn, and it just being plug and play you see. I also want to use it haha

albedozero, 2021-07-07:
No worries - you should be able to use any graphical text editor from the desktop (not sure what it is on Ubuntu, probably something that sounds notepad-ish), or here's a one-liner that should work:
```
sed -i "/shutdowntimer = t/d" headlesspi.py
```

Ashguitargolf, 2021-07-07:
I entered that line in terminal, changed patch, it still turned off. I want to recall the script in terminal to take out the shutdown command. I don't know how to do that in terminal you see.

albedozero, 2021-07-07:
If you want to edit in the terminal, the `pico` editor is pretty easy:
```
pico headlesspi.py
```
Control-s saves and Control-x exits.

That `sed` command you entered should have already deleted the offending line, however. Maybe this isn't the problem after all - are you seeing the message `Shutting down..` from the script when it happens? If not, maybe something else is causing it..

Ashguitargolf, 2021-07-07:
Well that sed line I put in must of done the trick. I checked the script and the shutdown command was gone. I exited, and now I'm changing patches and it works. WE (You) did it, for now. Thanks for your time Bill. I've been missing the states, I lived in Boise Idaho for 6 years up until 2019. So greetings from England and we'll no doubt cross paths on one forum or another. Cheers

Ashguitargolf, 2021-07-07:
I'm having fun putting my Behringer TD-3 through it, some really cool strange rhythms and harmonies

albedozero, 2021-07-07:
Nice! Glad I could help. I'm jealous of your transplanting to England - really enjoyed my trip there a few years ago. See you down the road!

closed 2021-07-07 by albedozero

Stooovie, 2023-05-25:
When I delete or #comment out the SHUTDOWN_BTN line, I get 

```
Traceback (most recent call last):
  File "/home/pi/./headlesspi.py", line 177, in <module>
    mainapp = HeadlessSynth()
  File "/home/pi/./headlesspi.py", line 105, in __init__
    self.load_bank(pxr.currentbank)
  File "/home/pi/./headlesspi.py", line 132, in load_bank
    self.select_patch(0, force=True)
  File "/home/pi/./headlesspi.py", line 142, in select_patch
    connect_controls()
  File "/home/pi/./headlesspi.py", line 43, in connect_controls
    if SHUTDOWN_BTN != None:
NameError: name 'SHUTDOWN_BTN' is not defined
```
The Pi is shutting down on program change. What else can I do? The power supply is a 2.4A and only the MIDI keyboard and USB audio dongle are connected to the Pi 4.

albedozero, 2023-05-25:
Sorry this isn't more clear - if you don't want holding the INC_PATCH/DEC_PATCH buttons to shut down the Pi, set SHUTDOWN_BTN to a different control change # (an unused one if you want to disable it). Commenting it out just breaks the code, as you've seen.

Stooovie, 2023-05-25:
> set SHUTDOWN_BTN to a different control change # (an unused one if you want to disable it).

This is the way. Fixed now. Thank you!