GeekFunkLabs/fluidpatcher #22: libfluidsynth package doesnt seem to be available on Ubuntu focal 20.04

clendee, 2021-11-07:
Hi,

I tried following the install instructions, mainly to put fluidpatcher on my Ubuntu focal 20.04 desktop so that I can try to push patches to my Pi running the synth software.

sudo apt install libfluidsynth

I get the error:
E: Unable to locate package libfluidsynth

I notice libfluidsynth2 is available for my distribution but once I try to open the patcher python starts coughing up errors.

albedozero, 2021-11-07:
Interesting - I haven't seen a distro with libfluidsynth2 available yet, and it's also strange to see 2 available but not 1. Your best bet is probably to just build fluidsynth from source on Ubuntu - it's quite easy with the cmake system they have set up. You can follow the instructions in the [FluidSynth wiki](https://github.com/FluidSynth/fluidsynth/wiki/BuildingWithCMake), or you can use the [squishbox-install.bash](https://github.com/albedozero/fluidpatcher/blob/master/assets/squishbox-install.bash) script - just comment out the [line](https://github.com/albedozero/fluidpatcher/blob/20294fd461f3d5d0c55930be27c7abe8ca5300b4/assets/squishbox-install.bash#L211) that tries to install libfluidsynth1

clendee, 2021-11-08:
Hi,
Here is a screenshot of libfluidsynth2 available in focal (20.04) from the Ubuntu Packages site.  I was searching for libfluidsynth but only libfluidsynth2 appears.

[image]

albedozero, 2021-11-08:
In any case, I'm thinking the instructions and install script should just use `fluidsynth` as a target, and not `libfluidsynth` since the former will point at the newest available version. It's good to see the newer versions of Debian/Ubuntu have switched to fluidsynth 2, as it's definitely better. If you do post the Python errors, would you also be able to post the results of

```
sudo apt-cache policy libfluidsynth2
```
or
```
sudo apt-cache policy fluidsynth
```
This way I can see if I created some version-specific bug. There's been a lot of development going on over there.

clendee, 2021-11-17:
Apologies for not getting back to you on this.  Some life stuff gets in the way sometimes, but I promise to follow it up.

albedozero, 2021-11-22:
I added some methods in a [recent commit](https://github.com/albedozero/fluidpatcher/commit/e0580682913ef5c0e528949ba94b47d59f141da1) that actually query the running instance of fluidsynth for its version number, so I can work around some of the differences in recent versions :) This should prevent code barfs if the most recent version of fluidsynth isn't installed. The install script has also been updated to use better repo targets in light of the recent jump to Debian 11.

closed 2021-11-22 by albedozero