GeekFunkLabs/fluidpatcher #23: No output on the screen (of the build kit)

clendee, 2021-11-17:
Hi,

Im sorry to be a thorn in your side.  Thank you for sending the build kit over, I have connected it all together and have a bit of troubleshooting to go through.

I went through the install instructions as per your website.  When I ran the script (the pipe to bash one), I noticed some errors during the running of it.  The first thing I noticed is that php7.4 was installed with RaspberryPi OS.  I downloaded your script code and changed every instance of 7.3 -> 7.4.  It then ran without errors.

Also, unfortunately Im getting no characters appear on the screen.  It is getting power.
Do you have any ideas on how I might troubleshoot this?

Thank you for your time.

albedozero, 2021-11-17:
No apologies needed - I appreciate your patience and help tracking down my mistakes :)

First, to enable the sound card edit your _/boot/config.txt_ file and add the following at the end:
```
dtoverlay=hifiberry-dac
```

I used to do this in the install script, but I had to remove it for compatibility and forgot to add it to the assembly instructions. Reboot, then re-run the install script and it will detect the sound card as hifiberry. Also, I did an update last night that introduced an LCD bug that I just fixed early this morning, so hopefully if you update the fluidpatcher code when running the script it will fix that.

Raspberry Pi OS just got updated to Debian 11 (bullseye), which is great but clearly I need to change some of the install targets. Thanks for figuring that out for me. Don't hesitate to write back if modifying _config.txt_ as above and re-running the install script doesn't fix your issues!

clendee, 2021-11-18:
Hi mate,

Thanks for your time and effort.  I added the `dtoverlay=hifiberry-dac` to the _/boot/config.txt_ and re-ran the install script as you suggested.  Should I comment out other `dtoverlay=` lines in _/boot/config.txt_ ?  

The output of dmesg is as follows:

[image]

Unfortunately I still have no output to the screen.

albedozero, 2021-11-18:
Ah, good to know the sound is working. You don't need to comment out the other `dtoverlay` lines - in fact it's probably better to leave them alone.

On the display, here are some checks to run through:
1. Does it light up at all? I test the LCDs before shipping, so unless it got damaged en route it should still get power.
2. Did you use a cross-head screwdriver to adjust the constrast potentiometer through its entire range while the unit is on? I usually have to do this before I can see anything.
3. If the above don't fix it, paste a copy of your _utils/hw_overlay.py_ file here - maybe something got messed up in the updates.

If it turns out there's something wrong with the LCD I'll get a new one to you.

albedozero, 2021-11-18:
Oops! My fault again - the v4 PCB is not out yet. I accidentally committed this upcoming version to the main repository. If you just delete the last section in _hw_overlay.py_ so the v3 values get used everything should work fine.

clendee, 2021-11-19:
Ahh, amazing.   Thanks so much.
Looking forward to playing around with this device. [And secretly delighted that the issue isn't my soldering skills]

[image]

closed 2021-11-19 by albedozero

Wapata, 2021-11-19:
Hi there !
Same problem here but I don't own " utils/hw_overlay.py " on my system. It seem that I did something wrong :-/

clendee, 2021-11-19:
Can you see the utils/hw_overlay.py file?  
Who owns it?  Thats a bit weird that the pi user doesn't own it.

I would probably just run the script again from scratch using the 

`curl -L git.io/squishbox | bash`

OR you can use the `

sudo chown pi:pi /home/pi/ -R`

This will change the owner and group of every file in the home directory to the pi user.

Wapata, 2021-11-19:
I did run the scrip again and tadaaaam it's working !
Midi input ok
Sound output ok
Screen output ok

I'll see the rest tomorrow, thank for the helping hand !

clendee, 2021-11-19:
Awesome, great stuff.

albedozero, 2021-11-19:
If the install script were initially run as root (i.e. using `sudo`) that might explain it.

> `curl -L git.io/squishbox | bash`

This actually won't fix things since the install script specifically avoids overwriting an existing _hw_overlay.py_ file, so that it doesn't mess up your hardware pin assignments for your specific build. It has to be edited manually (changing ownership with `sudo` if necessary) - [this README](https://github.com/albedozero/fluidpatcher/blob/master/utils/README.md) should help if needed.