GeekFunkLabs/fluidpatcher #28: Playing pad style presets can cause the Pi to cut out

clendee, 2021-11-21:
Hi,

I notice when playing 'Analog Pad' or other pad style presets, perhaps using too many fingers, ie pressing too many keys, the Pi will cut out.  I thought the Pi 3B+ should be enough to handle that amount of throughput (or am I looking at this the wrong way).

Is there a way to limit the polyphony on certain presents to prevent this from happening?

albedozero, 2021-11-22:
I have to say I'm impressed you're able to hold that many keys at once. I had to switch my sustain pedal to toggle and hold down about 5 octaves worth of notes to get the audio to cut out. I've tested this before and thought it would be an edge case, but another user encountered the same problem. The solution is to set `synth.cpu-cores` to `4` and `synth.polyphony` to `128`. FluidSynth drops the oldest notes when running out of voices, and these values seem to be sufficient that the processor doesn't max out and cut the audio. You can add these settings to your _SquishBox/squishboxconf.yaml_ file - check the [latest version](https://github.com/albedozero/fluidpatcher/blob/master/SquishBox/squishboxconf.yaml) in the repository to see how.

I added some tweaks to the installer to [allow realtime audio](https://jackaudio.org/faq/linux_rt_config.html), which deals with the `Failed to set thread to high priority` messages. `squishbox.py` isn't running multiple times - the green entries in `htop` are threads spawned by the original process. FluidSynth spawns a few, the GPIO library has one, etc.

I experimented with using `jackd2` a few months back and had lots of problems with crashing and segmentation faults, but it seems to be working now, so I updated the installer to use that as well!

closed 2021-11-22 by albedozero

clendee, 2021-11-22:
Thanks for your comments and your willingness to explore the questions and ideas.

Now that I have built the Squishbox, I can get more stuck in to its capabilities.  Here is my rendition of it.  A friend of mine has a laser cutter and built the box out of 3mm board.  Its quite study and fits neatly.  My friend said he'd be happy to share his design for any interested parties.
[image]

I feel that when I tried a different power supply, I was having less issues with the cutting out of various patches, so perhaps it was power related.