GeekFunkLabs/fluidpatcher #33: Crash when changing patches

RodMac, 2022-01-19:
Hi there - nice project you have here. Getting consistent crashes with headlesspi.py when changing patches using midi keyboard transport keys.

The setup I have is:
- Raspberry Pi 4 Model B Rev 1.1 with freshly installed Bullseye
- m-audio Axiom 49 Gen 1 midi keyboard connected via USB (powered via adapter). 
- PiSound HAT sound card

headlesspi.py launches successfully and plays notes like a normal piano. Changing banks works fine using the midi keyboard's Play button (transport keys).

But using the keyboard's Fast Forward and Rewind buttons (transport keys) to change patches consistently causes the Pi to shutdown. See screenshot.

[image]

checked /var/log/messages & /var/log/syslog without much to see (excerpts attached)

[image]
[image]

Faeb35, 2022-01-21:
Today I just installed fluidpatcher on an old Raspberry Pi 2. I really like it, it's an awesome software and does a great job! Big thanks @albedozero, I really like your software!!

I face the same issue as @RodMac does. I'm also using an Axiom 49 too, could also be first generation, I'm 
quite sure it's pretty old.
When I use one of the buttons to change the banks/patches, after about 5 seconds, the red LED starts to blink 10 times and the raspberry pi seams to shutdown (USB devices are powered off).
I'm guessing that using one of these bottons is "like constantly pressing it" and somehow triggers the shutdown routine, like demonstrated here:
https://www.youtube.com/watch?v=ONuXCzcuj7E&t=230s
So it seams to me, that it's not really a crash, rather an unintended shutdown of the Raspberry Pi. 

I also recognized the text "Shuttin down.." in the first screenshot of @RodMac, which seams to come from headlesspi.py, line 104, just before the line _subprocess.run('sudo shutdown -h now'.split())_.
After commenting this line out in the code, the red LED still starts blinking, but it raspberry pi did not shutdown anymore. So I set a # on the beginning of the lines 97 to 106, which disabled the shutdown routine for now to workaround the issue.

albedozero, 2022-01-21:
@Faeb35 you are correct that the "Shutting down..." message indicates that the headlesspi.py script thinks you're holding the patch button down, and is thus triggering an unintended shutdown. Thanks for the suggestion of a workaround - an even shorter one would be to comment [this single line](https://github.com/albedozero/fluidpatcher/blob/cf0e314bd4bdb0b19dbafc85b087268d830c75c5/headlesspi.py#L142), which would also stop the blinking LED.

However, this means that you have to unplug the Pi to shut it down, which is not usually recommended as you can corrupt your SD card. A lot of controllers use those transport buttons (play, record, rewind, etc.) to send MMC (multimedia control messages), which only send "on" values - never "off" - so if you can use one of the other buttons (f10-f18?) to switch patches you would still be able to trigger a shutdown when needed. Kudos for finding your own workaround!

Faeb35, 2022-01-23:
@albedozero thanks for your fast replay and providing all the information!
As I really wanted to use the fast forward buttons on my Axiom to change patches, I came up with another solution to initiate the shutdown, which I provided with pull request #34.

RodMac, 2022-01-24:
Thanks for your quick reply @albedozero and pointing me to the Troubleshooting which, I now see, already has solutions / work arounds for this scenario.
 
@Faeb35  I was able to change the transport keys to be momentary using the programming aspects of the Axiom keyboard and this solved the problem. I haven't fully deciphered the Axiom's manual and required button combinations to program it but for my keyboard (Axiom 49 gen 1) to change the transport buttons to momentary is something along the lines of:
- press the button to edit (eg the fastforward >> button)
- press the control assign button
- type 146 on the numeric keyboard which sets the button to "MIDI CC (on/off)" ie momentary
- Press the Data 1 button
- enter cc code for the button (eg 20)

closed 2022-01-25 by RodMac