GeekFunkLabs/fluidpatcher #37: (Question) Jack as an option on Fluidsynth audio.driver - RPi 3, Headphones

elbolutony, 2022-02-18:
Hi, first, thanks for you work on this. (En is not my native language, sorry in advance)
I've found/feel that the "real" latency is much less noticeable using jack as an audio driver than alsa.
Thing is that using jack won't autostart fluidsynth on boot.
It's possible? 
Secondary question:
I'm unable to effectively modify the period one ./jackrc; (or whatever i do on fluidsettings.xml)
"fluidsynth warning requested a period size of 64 got 444 instead" but i'll keep tinkering this, 

Thanks!

albedozero, 2022-02-19:
JACK should be started automatically by FluidSynth on boot. It seems the headphone jack cannot be used with a period size of less than 444, for technical reasons I don't completely understand. You can get lower with a USB sound card or sound card that connects to the GPIO.

Can you share the contents of your /home/pi/.jackdrc file and your /home/pi/SquishBox/squishboxconf.yaml file? That might help to figure out the issue.

albedozero, 2022-02-19:
The lines
```
JACK server starting in realtime mode with priority 10
creating alsa driver ... hw:Headphones|-|512|3|48000|0|2|nomon|swmeter|soft-mode|16bit
```
show that the JACK server is indeed being started by FluidSynth. JACK uses ALSA as a back-end driver, but it is running and enforcing lower-latency audio. Are you getting sound output?

The fluidsynth warnings about `Unknown integer parameter 'synth.sample-rate'` are because the sample rate should be a float: 48000.0 for example. This isn't breaking anything - fluidsynth adjusts its sample rate to match the 48000 you requested from JACK.

elbolutony, 2022-02-19:
It runs ok, sounds ok, but not autostarting on boot, i have to run it manually.

elbolutony, 2022-02-19:
Ok, i took your comment;

This is free software, and you are welcome to redistribute it
under certain conditions; see the file COPYING for details
JACK server starting in realtime mode with priority 10
self-connect-mode is "Don't restrict self connect requests"
creating alsa driver ... hw:Headphones|-|480|2|48000|0|2|nomon|swmeter|soft-mode|16bit
configuring for 48000Hz, period = 480 frames (10.0 ms), buffer = 2 periods
ALSA: final selected sample format for playback: 16bit little-endian
ALSA: use 2 periods for playback
fluidsynth: Connection of Midi Through Port-0 succeeded
fluidsynth: Connection of K61P MIDI 1 succeeded
Connection is already subscribed
Connection is already subscribed
Loading bank 'bank1.yaml' ..
fluidsynth: warning: Sample 'SineWave': ROM sample ignored
Bank loaded.
Selected patch 1/26: Piano

i managed to get this without noticeable xruns (except on the start).

albedozero, 2022-02-19:
Did you install using `curl -L git.io/squishbox | bash`? What happens when you enter
```
sudo systemctl status squishbox.service
```

elbolutony, 2022-02-19:
pi@raspberrypi:~ $ sudo systemctl status squishbox.service
● squishbox.service - Headless Pi Synth
     Loaded: loaded (/etc/systemd/system/squishbox.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-02-19 03:05:20 GMT; 9min ago
   Main PID: 394 (python3)
      Tasks: 8 (limit: 1597)
        CPU: 4.624s
     CGroup: /system.slice/squishbox.service
             └─394 python3 /home/pi/headlesspi.py

Feb 19 03:12:49 raspberrypi sudo[906]: pam_unix(sudo:session): session closed for user root
Feb 19 03:12:50 raspberrypi sudo[909]:       pi : PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/tee /sys/class/leds/led0/brightness
Feb 19 03:12:50 raspberrypi sudo[909]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Feb 19 03:12:50 raspberrypi sudo[909]: pam_unix(sudo:session): session closed for user root
Feb 19 03:12:50 raspberrypi sudo[912]:       pi : PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/tee /sys/class/leds/led0/brightness
Feb 19 03:12:50 raspberrypi sudo[912]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Feb 19 03:12:50 raspberrypi sudo[912]: pam_unix(sudo:session): session closed for user root
Feb 19 03:12:50 raspberrypi sudo[915]:       pi : PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/tee /sys/class/leds/led0/brightness
Feb 19 03:12:50 raspberrypi sudo[915]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Feb 19 03:12:50 raspberrypi sudo[915]: pam_unix(sudo:session): session closed for user root

Not sound whatsoever.

albedozero, 2022-02-19:
Aha - it was a bug. [This line](https://github.com/albedozero/fluidpatcher/commit/815fa3c72fd953d0367e6c95d8f6b1e2ffeaf2cd) missing from the install script meant that JACK could not be started automatically. You should be able to fix this by running the install script again
```
curl -L git.io/squishbox | bash
```
Answer no/no change to all options except for "What script should be run on startup?" - answer headlesspi.py and it will overwrite the unit file and should work when you restart!

elbolutony, 2022-02-19:
Cool, now it's starting on boot! 
Magnificent. 
Thanks again, i'm closing the issue.

closed 2022-02-19 by elbolutony