GeekFunkLabs/fluidpatcher #38: No sound or lcd output

jik1317, 2022-04-08:
Hi,
I recently put together the v3 version of the sqiushbox with the pcb. (Litterally the day before v4 was released! aghhhhhhh! I might have to upgrade later, but I wanna get this one working first)
Anyways, I installed the firmware onto the raspberry pi and went through the setup, but I'm getting neither a sound output or text on the lcd.
I can confirm that the lcd works and it is reciving power, I also tested all the wire paths for continuity and they are fine. Perhaps its an issue with the pi's gpio?
I'm an Arduino guy so any help would be greatly appreciated and I will gladly provide more information/details if anyone can help me.

albedozero, 2022-04-08:
You'll need to edit the /home/pi/utils/hw_overlay.py file, uncomment the section that refers to "`PCB v4`", and delete the rest. I'll need to update the assembly instructions to reflect this 🤦.

That should get the LCD working, as long as you've adjusted the contrast screw so that it's not completely dark. On the no sound issue, did you definitely add `dtoverlay=hifiberry-dac` to _/boot/config.txt_, and you're sure you gave the `sndrpihifiberry` answer to the audio outputs question in the setup script?

jik1317, 2022-04-08:
That worked great and solved both issues. Thank you so much!

closed 2022-04-08 by jik1317