GeekFunkLabs/fluidpatcher #44: Question about routing

kentron, 2022-05-22:
Hi Albedozero, how are you

Thank you for this wrapper, I'm currently adapting it for a performance with an early music ensemble. I intend to emulate pipe organ stops, but I wonder what it would be the best way to do so. I'm trying to avoid assigning volume = 0 for a particular channel. Instead, I would like to turn on/off a given program by routing/unrouting channel 1 to the channel assigned in the bank file for the preset. The behavior should be like turning on and off a particular preset. 
Is there a way to unroute a previously routed channel? Do I Need to extend fswrap.py to  add a function to wrap fluidsynth's api call 'delete_fluid_midi_router_rule'? In that case: how do I refer to the id of an existing rule?

Thank you very much for the advice

albedozero, 2022-05-22:
Unfortunately the only way to remove a rule once it's added to the router is to clear all the rules and start again. As the [FluidSynth API states](https://www.fluidsynth.org/api/group__midi__router.html#gafa33ca65b0db6aaf16627c1972edaeb3), once you pass a rule to `fluid_midi_router_add_rule()` it should not be accessed again. This is because rules are stored in the router in a linked-list structure - removing a rule would break the link and probably cause a segfault. I think `delete_fluid_midi_router_rule()` is meant to be used on a rule you've created but not yet added to the router.

What's your reason for not wanting to use CC7=0 to mute particular channels? I know that this can cause unwanted behavior if you also use the volume slider on your keyboard to control the volume on all channels - those muted voices will pop back on if you move the slider. The workaround is to use CC11 to mute voices when needed as discussed in [FluidPatcher Lesson 4](https://youtu.be/K_TKv0kQubU). The sound volume is the _product_ of MIDI volume (CC7) and Expression (CC11), so an expression level of 0 is unaffected by changing the volume. This was designed into the MIDI specification for pretty much this exact reason. Here's an example bank file that would do something like you're suggesting:

```yaml
patches:
  Drawbars:
    1: organ/Oberheim OB-3.sf2:000:001
    2: organ/Oberheim OB-3.sf2:000:002
    3: organ/Oberheim OB-3.sf2:000:003
    router_rules:
    - {type: note, chan: 1=2-3}
    - {type: cc, chan: 1, par1: 13=11}
    - {type: cc, chan: 1=2, par1: 14=11}
    - {type: cc, chan: 1=3, par1: 15=11}
```

kentron, 2022-05-25:
Hi, thank you very much for your detailed response. I wasn't aware that fluidsynth stored routing information in a r/o structure. This changes the plan for sure.

I've red in some forum (https://www.midi.org/forum/8492-is-there-a-channel-mute-message,-and-if-so-is-it-sysex-or-a-cc) some objections about using the volume control CC7 to mute a voice, but I didn't have the time to actually test if there is any issue related with their claim. 
Anyway, when I see the behavior of many sequencers, it looks like the mute button doesn't affect the volume of a particular channel. Also, I didn't find a way to query the synth to tell what is the actual CC7 value for a channel before muting, in order to restore it when reactivating the voice.... Would it be a possible alternative to use fluid_synth_unset_program/fluid_synth_program_select instead?
I was thinking to add a route rule poiting to a listener which will query fluidsynth for information about the channel, then if the program is assigned it means the state is on, otherwise off. 

I don't know if that could cause any lag, since the activation/deactivation of a voice needs to be immediate. Sadly I'm out of time this week to try all this but I look forward in seven days!..

Thank you again,

albedozero, 2022-05-25:
I see in that forum post that someone was worried about losing polyphony to voices that are being played at zero volume. That may or may not be true for the Roland Sound Canvas, but FluidSynth allocates only the voices it needs to produce sound, and won't use them up on channels that are at zero volume, even if those channels receive notes. I think using CC 11 routed to some toggling pads/buttons you your keyboard to mute the channels, and CC 7 for overall volume as in my (edited) example above would give you the smoothest operation. Another way to do it would be to create individual patches for each combination of organ stops and switch between them - you can use a router rule with a `patch` parameter as described in the [bank file doc](https://github.com/albedozero/fluidpatcher/blob/master/patcher/file_formats.md#bank-files) to select patches directly with a button or key.

The functions fluid_synth_unset_program/fluid_synth_program_select are [already wrapped](https://github.com/albedozero/fluidpatcher/blob/1f3a5aaaf631c3f21792ac506b172e9fa0622af1/patcher/fswrap.py#L716), but they aren't really meant to be accessed through the FluidPatcher API. The whole point of FluidPatcher is to create patches with the programs you want on each channel, then switch between the patches. You could add code to one of the [listener](https://github.com/albedozero/fluidpatcher/blob/1f3a5aaaf631c3f21792ac506b172e9fa0622af1/squishbox.py#L147) functions that would handle a message from your custom router rule to set/unset programs on a given channel, but there are already ways to accomplish the same effect using bank files, without having to go into the code.

kentron, 2022-05-27:
Hi AlbedoZero, thank you again for the detailed information about fluidsynth internals. While I am still not quite sure how to implement this purely using the bank files, for a current solution I'm using a combination of program change and patches, which add specific programs from different pipe organ sounfonts , creating a mix. 

Ill try to think on an implementation to emulate the way a pipe organ behaves. Perhaps some bank files can have a directive at the init level, to separate banks and its rules from the 'normal' banks. I'll write you later 👍

closed 2022-05-27 by kentron