GeekFunkLabs/fluidpatcher #55: unable to update squishbox from version 0.7.9

albedozero, 2022-08-07:
Hi, I'm not able to update squishbox to 0.7.11 version from 0.7.9 from software option.

_Originally posted by @tizfra in https://github.com/albedozero/fluidpatcher/discussions/54#discussioncomment-3344630_

albedozero, 2022-08-10:
Thanks - there was a bug in the squishbox.py script that is fixed in version 0.7.12 where double-digit version numbers were not being compared correctly. To update from an earlier version you will need to run the install/update script from a command line with
```bash
curl -L git.io/squishbox | bash
```
You can then choose to install/update FluidPatcher and say no to the other options. This is always a fallback procedure if the squishbox.py update fails.

closed 2022-08-10 by albedozero

tizfra, 2022-08-11:
Ok, I've been able to update squishbox, but now it can't sound any patch.
In system menu i can see that midi signal is ok, but no sound comes.