GeekFunkLabs/fluidpatcher #69: RFE: adding all instruments from FluidR3_GM.sf2

zenfish, 2022-12-14:
First - thanks! The Squishbox works like a champ (may send notes of some minor issues, but for now, gr8) I'd just set up a raspberry with similar functionality, but your little box brings a much nicer physical interface to the metaphorical table.

Secondly - noticed you used the ubiquitous FluidR3_GM.sf2 sound font... there's an example at:

https://gist.github.com/albedozero/721442c1d112839c71fd8f7d9ce1a98b

It runs through the full number (close to 200 different instruments)... if you simply put the file (any reasonable file name ending in .yaml seems to work) in "SquishBox/banks", your basic "router_rules" at the top of the file, it works like a champ. You might consider putting it as one of your sample bank files, but of course YMMV... scrolling through even 200 patches doesn't take that long, and it's all the more impressive when kicking the tires of the squishbox.

You may have this somewhere in your notes/dox, but I didn't see it.

Tnx again!

Stooovie, 2023-05-24:
No matter what I do, doesn't work for me. I did copy the router_rules section, and the init section on the bottom, i did put the proper filename into squishboxconf.yaml, yet upon reboot it just doesn't play. The file looks like this and produces no sound at all at any channel or patch number:
```

router_rules:
  - {type: cc, chan: 1=2-16, par1: 7}
  - {type: cc, chan: 1=2-6, par1: 1}
  - {type: cc, chan: 1=2-6, par1: 64}
  - {type: cc, chan: 1=2-6, par1: 91}
  - {type: pbend, chan: 1=2-6}

patches:
  Yamaha Grand Piano:
    1: FluidR3_GM.sf2:000:000
  Bright Yamaha Grand:
    1: FluidR3_GM.sf2:000:001
  Electric Piano:
    1: FluidR3_GM.sf2:000:002
  Honky Tonk:
    1: FluidR3_GM.sf2:000:003
  Rhodes EP:
    1: FluidR3_GM.sf2:000:004
  Legend EP 2:
    1: FluidR3_GM.sf2:000:005
  Harpsichord:
    1: FluidR3_GM.sf2:000:006
  Clavinet:
    1: FluidR3_GM.sf2:000:007
  Celesta:
    1: FluidR3_GM.sf2:000:008
  Glockenspiel:
    1: FluidR3_GM.sf2:000:009
  Music Box:
    1: FluidR3_GM.sf2:000:010
  Vibraphone:
    1: FluidR3_GM.sf2:000:011
  Marimba:
    1: FluidR3_GM.sf2:000:012
  Xylophone:
    1: FluidR3_GM.sf2:000:013
  Tubular Bells:
    1: FluidR3_GM.sf2:000:014
  Dulcimer:
    1: FluidR3_GM.sf2:000:015
  DrawbarOrgan:
    1: FluidR3_GM.sf2:000:016
  Percussive Organ:
    1: FluidR3_GM.sf2:000:017
  Rock Organ:
    1: FluidR3_GM.sf2:000:018
  Church Organ:
    1: FluidR3_GM.sf2:000:019
  Reed Organ:
    1: FluidR3_GM.sf2:000:020
  Accordian:
    1: FluidR3_GM.sf2:000:021
  Harmonica:
    1: FluidR3_GM.sf2:000:022
  Bandoneon:
    1: FluidR3_GM.sf2:000:023
  Nylon String Guitar:
    1: FluidR3_GM.sf2:000:024
  Steel String Guitar:
    1: FluidR3_GM.sf2:000:025
  Jazz Guitar:
    1: FluidR3_GM.sf2:000:026
  Clean Guitar:
    1: FluidR3_GM.sf2:000:027
  Palm Muted Guitar:
    1: FluidR3_GM.sf2:000:028
  Overdrive Guitar:
[... 302 lines cut ...]
    1: FluidR3_GM.sf2:128:032
  Jazz 1:
    1: FluidR3_GM.sf2:128:033
  Jazz 2:
    1: FluidR3_GM.sf2:128:034
  Jazz 3:
    1: FluidR3_GM.sf2:128:035
  Jazz 4:
    1: FluidR3_GM.sf2:128:036
  Brush:
    1: FluidR3_GM.sf2:128:040
  Brush 1:
    1: FluidR3_GM.sf2:128:041
  Brush 2:
    1: FluidR3_GM.sf2:128:042
  Orchestra Kit:
    1: FluidR3_GM.sf2:128:048
	
init:
  messages: [cc:4:37:60, cc:4:65:127, cc:4:68:127,
             cc:5:71:0, cc:5:73:0, cc:5:74:0, cc:5:78:40, cc:5:79:40, cc:5:80:80, cc:5:81:127, cc:5:82:80, 
             cc:5:71:0, cc:5:73:0, cc:5:74:0, cc:5:78:40, cc:5:79:40, cc:5:80:80, cc:5:81:127, cc:5:82:80, 
             cc:6:100:0, cc:6:101:0, cc:6:6:12, cc:6:38:0]
```
I don't see any functional difference to the default bank#.yaml files, yet those play and this doesn't. Any clues?

albedozero, 2023-05-24:
see my comment on #78

albedozero, 2023-11-18:
I know it's been a bit but trying to clean up some issues. The ability to scroll through all the presets in FluidR3_GM.sf2 - or any soundfont you have loaded on to the SquishBox - is provided by the "Open Soundfont" option in the menu. Further, you can "Add as Patch" any of the presets in the font as you scroll through presets, providing a nice way to try out sounds and build up a bank without even having to do any text file editing.

If more functionality is desired feel free to reopen.

closed 2023-11-18 by albedozero