GeekFunkLabs/fluidpatcher #74: Unable to start FluidPather (pyw) after running Windows installer due to failing wxpython pip install

jawn, 2023-02-19:
Note: I am not too familiar with python issues like below, feel free to tell me to ask the question on stackoverflow.com if that's a better idea.

## Steps to reproduce:
1. installed the latest x64 exe installer from https://github.com/albedozero/fluidpatcher/releases/tag/v0.7.15
- Python was already installed before.
2. Start the desktop shortcut FluidPatcher (corresponds to pyw).

## Expected result:
- FluidPatcher starts.
- In addition, installer should catch error when Python pip installation fails. Now it completes.

## Actual result:
- FluidPatcher doesn't start, no error message.

Troubleshooting:
- Re-ran the installer and isolated the issue to the following installer step:
# Succeeds:
Execute: py -3.10-64 -m pip install -U oyaml
# Succeeds:
Execute: py -3.10-64 -m pip install -U mido
# Fails in seperate concole windows, 
Execute: py -3.10-64 -m pip install -U wxpython
Execute: py -3.10-64 -m pip install -U --pre -f https://wxpython.org/Phoenix/snapshot-builds/ wxpython

- Re-ran pip commands:
For oyaml, mido: succeeds
```
py -3.10-64 -m pip install -U mido
Requirement already satisfied: mido in c:\python311\lib\site-packages (1.2.10)
```

For wxpython: fails
```
py -3.10-64 -m pip install -U wxpython
Collecting wxpython
  Using cached wxPython-4.2.0.tar.gz (71.0 MB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\Bernard\AppData\Local\Temp\pip-install-dokcizpt\wxpython_662eefb4314c47eba7b194b4d07a8e18\setup.py", line 27, in <module>
          from buildtools.config import Config, msg, opj, runcmd, canGetSOName, getSOName
        File "C:\Users\Bernard\AppData\Local\Temp\pip-install-dokcizpt\wxpython_662eefb4314c47eba7b194b4d07a8e18\buildtools\config.py", line 30, in <module>
          from attrdict import AttrDict
      ModuleNotFoundError: No module named 'attrdict'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

```
**Background Info**
Let us know:
- Which script you are using: PYW shortcut created by Windows intaller
- Your platform: Windows
- FluidSynth version (type `fluidsynth --version` at a command line)
C:\ProgramData\chocolatey\bin\fluidsynth.exe v2.13 (installed as chocolatey package)
C:\Program Files\FluidPatcher\fluidsynth.exe v2.3.0 
- any other relevant info (OS version, etc.) as you can
Windows 10 22H2 x64
Python 3.1.0

jawn, 2023-02-19:
This is the cause https://github.com/wxWidgets/Phoenix/issues/2225
Recommendation seems to be add an extra package install: attrdict3

albedozero, 2023-03-12:
It took me a while to get anywhere on this one. I did fix a bug related to your issue, but I can't duplicate your wxpython install problem. The installer wasn't properly checking the output of `py -3.10-64 -VV`, and would proceed even if Python 3.10 wasn't installed. I updated this behavior in the Windows installers in [release 0.7.15](https://github.com/albedozero/fluidpatcher/releases/tag/v0.7.15).

However, I can't duplicate your error (or any of what's described in https://github.com/wxWidgets/Phoenix/issues/2225, though I have no desire to drop into that scrum) when running `py -3.10-64 -m pip install -U wxpython` - it installs fine for me even on a system with no other python versions or extra packages:
```
C:\Users\BillyP>py -0
Installed Pythons found by py Launcher for Windows
 -3.10-64 *

C:\Users\BillyP>pip list
Package    Version
---------- -------
pip        23.0.1
setuptools 65.5.0

C:\Users\BillyP>py -3.10-64 -m pip install -U wxpython
Collecting wxpython
  Using cached wxPython-4.2.0-cp310-cp310-win_amd64.whl (18.0 MB)
Collecting pillow
  Using cached Pillow-9.4.0-cp310-cp310-win_amd64.whl (2.5 MB)
Collecting six
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting numpy
  Using cached numpy-1.24.2-cp310-cp310-win_amd64.whl (14.8 MB)
Installing collected packages: six, pillow, numpy, wxpython
Successfully installed numpy-1.24.2 pillow-9.4.0 six-1.16.0 wxpython-4.2.0
```
I notice when you installed mido it looked in what appears to be the Python 3.11 packages location:
> ```
> Requirement already satisfied: mido in c:\python311\lib\site-packages (1.2.10)
> ```
Maybe something is up with how the `py` command runs on some systems? The errors you and others are getting look like what happens when Python 3.10 is not installed -- wxPython won't work with Python 3.11 on Windows. However, `py -3.10-64` should complain if 3.10 isn't installed.

Side note - how did you debug my Windows installers to find the failing commands in the first place? Does NSIS have a debug mode I don't know about? I'm no expert and would appreciate the advice. Thanks for your help in any case.

closed 2023-03-12 by albedozero

jawn, 2023-03-13:
> Side note - how did you debug my Windows installers to find the failing commands in the first place? Does NSIS have a debug mode I don't know about? I'm no expert and would appreciate the advice. Thanks for your help in any case.

@albedozero to see the details, I use the **Show details** button in the installer.
That plus the cmd windows launched provided the extra details.

jawn, 2023-03-15:
**Update:** workaround was to indeed install Python 3.10, then the fluidpatcher install succeeds and can be started.

If 3.10 isn't installed, the installer checks for 3.10 and finds the 3.11 site packages assuming no action was needed.