GeekFunkLabs/fluidpatcher #78: Cannot use the fluid_gmbank.yaml bank

Stooovie, 2023-05-24:
No matter what I do, doesn't work for me. I did copy the `router_rules` section, and the `init` section on the bottom, i did put the proper filename into squishboxconf.yaml (currentbank: fluid_gmbank.yaml), paths are unchanged, yet upon reboot it just doesn't play. The file looks like this and produces no sound at all at any channel or patch number:

`router_rules:
  - {type: cc, chan: 1=2-16, par1: 7}
  - {type: cc, chan: 1=2-6, par1: 1}
  - {type: cc, chan: 1=2-6, par1: 64}
  - {type: cc, chan: 1=2-6, par1: 91}
  - {type: pbend, chan: 1=2-6}

patches:
  Yamaha Grand Piano:
    1: FluidR3_GM.sf2:000:000
  Bright Yamaha Grand:
    1: FluidR3_GM.sf2:000:001
  Electric Piano:
    1: FluidR3_GM.sf2:000:002
  Honky Tonk:
    1: FluidR3_GM.sf2:000:003
  Rhodes EP:
    1: FluidR3_GM.sf2:000:004
  Legend EP 2:
    1: FluidR3_GM.sf2:000:005
  Harpsichord:
    1: FluidR3_GM.sf2:000:006
  Clavinet:
    1: FluidR3_GM.sf2:000:007
  Celesta:
    1: FluidR3_GM.sf2:000:008
  Glockenspiel:
    1: FluidR3_GM.sf2:000:009
  Music Box:
    1: FluidR3_GM.sf2:000:010
  Vibraphone:
    1: FluidR3_GM.sf2:000:011
  Marimba:
    1: FluidR3_GM.sf2:000:012
  Xylophone:
    1: FluidR3_GM.sf2:000:013
  Tubular Bells:
    1: FluidR3_GM.sf2:000:014
  Dulcimer:
    1: FluidR3_GM.sf2:000:015
  DrawbarOrgan:
    1: FluidR3_GM.sf2:000:016
  Percussive Organ:
    1: FluidR3_GM.sf2:000:017
  Rock Organ:
    1: FluidR3_GM.sf2:000:018
  Church Organ:
    1: FluidR3_GM.sf2:000:019
  Reed Organ:
    1: FluidR3_GM.sf2:000:020
  Accordian:
    1: FluidR3_GM.sf2:000:021
  Harmonica:
    1: FluidR3_GM.sf2:000:022
  Bandoneon:
    1: FluidR3_GM.sf2:000:023
  Nylon String Guitar:
    1: FluidR3_GM.sf2:000:024
  Steel String Guitar:
    1: FluidR3_GM.sf2:000:025
  Jazz Guitar:
    1: FluidR3_GM.sf2:000:026
  Clean Guitar:
    1: FluidR3_GM.sf2:000:027
  Palm Muted Guitar:
    1: FluidR3_GM.sf2:000:028
  Overdrive Guitar:
    1: FluidR3_GM.sf2:000:029
[... 302 lines cut ...]
  Jazz 1:
    1: FluidR3_GM.sf2:128:033
  Jazz 2:
    1: FluidR3_GM.sf2:128:034
  Jazz 3:
    1: FluidR3_GM.sf2:128:035
  Jazz 4:
    1: FluidR3_GM.sf2:128:036
  Brush:
    1: FluidR3_GM.sf2:128:040
  Brush 1:
    1: FluidR3_GM.sf2:128:041
  Brush 2:
    1: FluidR3_GM.sf2:128:042
  Orchestra Kit:
    1: FluidR3_GM.sf2:128:048
	
init:
  messages: [cc:4:37:60, cc:4:65:127, cc:4:68:127,
             cc:5:71:0, cc:5:73:0, cc:5:74:0, cc:5:78:40, cc:5:79:40, cc:5:80:80, cc:5:81:127, cc:5:82:80, 
             cc:5:71:0, cc:5:73:0, cc:5:74:0, cc:5:78:40, cc:5:79:40, cc:5:80:80, cc:5:81:127, cc:5:82:80, 
             cc:6:100:0, cc:6:101:0, cc:6:6:12, cc:6:38:0]
`

I don't see any functional difference to the default bank#.yaml files, yet those play and this doesn't. Any clues?

albedozero, 2023-05-24:
When I run this in the GUI I get an error about a tab character at the start of line 390 (right where you added the `init` block). [According to the YAML spec](https://yaml.org/spec/1.2.2/#61-indentation-spaces), which bank files are based on, tabs can't be used at the beginning of lines.

You don't need that `init` block, or the `router_rules` block - _bank0.yaml_ uses the init block to set some initial control change values for instruments on different channels, and the router rules allow a keyboard sending notes on channel 1 to also talk to those other channels. The gmbank you're using just uses the default values for all instruments, so everything can be on channel 1.

Stooovie, 2023-05-24:
Still, even with your own fluid_gmbank.yaml from [https://gist.github.com/albedozero/721442c1d112839c71fd8f7d9ce1a98b](https://gist.github.com/albedozero/721442c1d112839c71fd8f7d9ce1a98b)
I'm not getting anything

albedozero, 2023-05-25:
Hmm. All the settings look good. Try
```
aconnect -iol
```
to see if the keyboard is being connected to FluidSynth, and 
```
speaker-test --device hw:Headphones
```
to confirm that sound is in fact working.

Stooovie, 2023-05-25:
aconnect -iol gives me

```
client 0: 'System' [type=kernel]
    0 'Timer           '
	Connecting To: 128:0
    1 'Announce        '
	Connecting To: 128:0
client 14: 'Midi Through' [type=kernel]
    0 'Midi Through Port-0'
	Connecting To: 128:0
client 20: 'Impact LX88+' [type=kernel,card=1]
    0 'Impact LX88+ MIDI1'
	Connecting To: 128:0
    1 'Impact LX88+ MIDI2'
	Connecting To: 128:0
client 128: 'FLUID Synth (396)' [type=user,pid=396]
    0 'Synth input port (396:0)'
	Connected From: 0:1, 14:0, 20:0, 20:1, 0:0
```

and the sound test does indeed play pink noise. This is 100% clean install. Rpi4 with 32bit Raspbian Bullseye with only the Fluidpatcher script ran after install. Only the Nektar keyboard and headphones are connected to the Pi.

Stooovie, 2023-05-25:
I can also play audio just fine after stopping the squishbox.service with
`fluidsynth -a pulseaudio -m alsa_seq -l -i sf2/GeneralUser.sf2 midi/elevatorgroove.mid `

But live playing does not produce any sound.

Stooovie, 2023-05-25:
Somehow, I have absolutely no idea how, I am now able to produce the basic patch on my LX88, but now I'm unable to change the patches. 

The LX88 emits control messages on Ch15 but notes on Ch0. Even when I change "Chan" in headlesspi.py to 15, I can not control the patches, only play notes with the first available patch. I am testing with aseqdump -p 20:0 and the numbers I put in headlesspi.py seem to be correct. 

The button I want to go to next patch is showing as this in aseqdump:
```
 20:0   Control change         15, controller 112, value 127
 20:0   Control change         15, controller 112, value 0
```
But when I set Chan to 15 and Controller to 112, and run ./headlesspi.py, pressing the button does nothing.

albedozero, 2023-05-25:
> Why does running ./headlesspi.py always say "Connection is already subscribed"? Even when I stop or even disable the squishbox service. That looks suspicious.

The warning is coming from fluidsynth. When fluidsynth autoconnects, the [fluid_alsa_seq_autoconnect](https://github.com/FluidSynth/fluidsynth/blob/7633d4aec77452e0184348d94ea13f13868633c2/src/drivers/fluid_alsa.c#LL1037C13-L1037C39) function goes through every port on your system and tries to connect it to FluidSynth, but if the port is already connected it skips it with this warning. One of them is the `system:announce` port, which is always connected to listen for new devices being plugged in, so maybe it's slightly slopping coding, but it doesn't cause a problem.

> when I set Chan to 15 and Controller to 112, and run ./headlesspi.py, pressing the button does nothing.

MIDI channels are numbered starting from 1 according to the MIDI spec. Fluidpatcher follows this rule, since pretty much all keyboards do as well. Unfortunately some programs like `aseqdump` and even FluidSynth itself number starting from 0.

TLDR your keyboard is sending notes on MIDI channel 1 and control changes on MIDI channel 16, so in _headlesspi.py_ you should set
```
CHAN=16
```

Stooovie, 2023-05-30:
Unfortunately no. I just tried. Aseqdump returns channel 0 on notes playing and channel 15 on controls like this:

```
 32:0   Note on                 0, note 48, velocity 43
 32:0   Note off                0, note 48
 32:0   Control change         15, controller 112, value 127
 32:0   Control change         15, controller 112, value 0
```

 but when I set Ch=15 and the appropriate controllers (112) like this:

```
CHAN = 15
TYPE = 'cc'             # 'cc' or 'note'
DEC_PATCH = 111          # decrement the patch number
INC_PATCH = 112
```
I can play notes but can not change patches. I have no idea why.

albedozero, 2023-05-30:
> but changing patches indeed DOES work when I set Ch to 16 (sixteen!)
> 
> `CHAN = 16`
> 
> Why is that? Is channel 16 some kind of global channel? Thank you!

`aseqdump` numbers MIDI channels starting from 0, which is not the accepted way. Almost all keyboards and the MIDI specification start numbering them at channel 1. So your keyboard is in fact sending control changes on MIDI channel 16 - `aseqdump` just reports it as 15.

Stooovie, 2023-05-30:
That's crazy :) Thank you, this one is solved.

Too bad SOME of the controls send on channel 0 (the rotary knobs) and the transport buttons send on channel 15, I'd love to be able to do incremental by the transport buttons and direct by the knobs, but that's just a minor inconvenience,