GeekFunkLabs/fluidpatcher #79: Doesn't install on macOS Ventura

Stooovie, 2023-05-24:
Running the provided commands results in 

```
Collecting oyaml
  Downloading oyaml-1.0-py2.py3-none-any.whl (3.0 kB)
Collecting wxpython
  Downloading wxPython-4.2.0.tar.gz (71.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 71.0/71.0 MB 26.3 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/private/tmp/pip-install-tmsupjic/wxpython_3c93365dc10b4964824c1aa6308d18d8/setup.py", line 27, in <module>
          from buildtools.config import Config, msg, opj, runcmd, canGetSOName, getSOName
        File "/private/tmp/pip-install-tmsupjic/wxpython_3c93365dc10b4964824c1aa6308d18d8/buildtools/config.py", line 30, in <module>
          from attrdict import AttrDict
      ModuleNotFoundError: No module named 'attrdict'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed
```
and

```
gcc -shared assets/patchcord.c -o patchcord.so && sudo mv -f patchcord.so /usr/lib/ladspa/

assets/patchcord.c:35:10: fatal error: 'ladspa.h' file not found
#include "ladspa.h"
         ^~~~~~~~~~
1 error generated.
```

Stooovie, 2023-05-24:
in Ubuntu it installs but trying to run python3 fluidpatcher.pyw results in `ModuleNotFoundError: no module called 'oyaml'`

albedozero, 2023-05-24:
The Mac OS X commands are my best guess as I don't currently have the ability to test on that platform. WxPython often takes a bit to catch up with new releases of Python and/or the OS version on different platforms, which is why I'm considering other options for a GUI. Some things you might try:

- On Windows, WxPython installs fine with older versions of Python - try installing in a `venv` with an earlier version of Python
- On Linux, because there are so many flavors it isn't possible to have a single package available in PyPI, so you have to search the package index for your platform. Try doing `brew install python3-wxgtk4.0` similar to the commands for Ubuntu, or search for a `wxgtk` package using `brew` commands

What happens when you do a `brew install ladspa-sdk`? If that works fine you should be able to build the patchcord LADSPA plugin, but it might be necessary to set some PATH variable on Mac OS or something. You don't actually need to do this step unless you want to use LADSPA effect plugins, and you would have to find or build these plugins for Mac OS as well.

Let me know if you get anywhere with some of these suggestions, and feel free to ask more questions

closed 2023-11-18 by albedozero