GeekFunkLabs/fluidpatcher #80: midi.autoconnect not working

Stooovie, 2023-05-25:
My goal is to have the Pi4 with Fluidsynth/Fluidpatcher permanently turned on and only having to turn on the MIDI keyboard (Nektar LX88+) to start playing.

Currently, if I disconnect/reconnect (or turn off/on) the keyboard, Fluidsynth doesn't start playing unless I manually restart the squishbox service. 

midi.autoconnect is already set to "1" in squishboxconf.yaml. 

What else can I do? Thanks!

albedozero, 2023-05-25:
This might be a fluidsynth problem with keyboards that present as multiple devices - one for keybed and one for controls; another user had a similar problem using a Keystation MK3 in #71. If you run fluidsynth from the command line with `-o midi.autoconnect=1` do you get the same behavior?

Stooovie, 2023-05-30:
I get this but still, no audio after turning the keyboard off and back again, despite this saying it's connected succesfully. 

```
pi@rpi:/usr/local/bin $ fluidsynth -o midi.autoconnect=1
FluidSynth runtime version 2.3.2
Copyright (C) 2000-2023 Peter Hanappe and others.
Distributed under the LGPL license.
SoundFont(R) is a registered trademark of Creative Technology Ltd.

fluidsynth: Connection of Midi Through Port-0 succeeded
fluidsynth: Connection of Impact LX88+ MIDI1 succeeded
fluidsynth: Connection of Impact LX88+ MIDI2 succeeded
fluidsynth: warning: Failed to set thread to high priority
Type 'help' for help topics.

> fluidsynth: warning: Failed to set thread to high priority
fluidsynth: Connection of Impact LX88+ MIDI1 succeeded
fluidsynth: Connection of Impact LX88+ MIDI2 succeeded
```
There is no audio until I call either `./headlesspi.py` or `sudo systemctl restart  squishbox.service`

Stooovie, 2023-05-30:
For posterity, if someone's in the same situation. I made a `nektar_connect.sh` script with just 
```
#!/bin/bash
sudo systemctl restart squishbox.service
```
and a new file in `/etc/udev/rules.d` called 99-nektar_connect.rules with this content:
`ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="2467", ATTRS{idProduct}=="2018", RUN+="/home/pi/nektar_connect.sh"`

I found the idVendor and idProduct numbers with lsusb.

Now, whenever I turn on the keyboard, it restarts the squishbox service, and everything works as expected.

closed 2023-05-30 by albedozero