GeekFunkLabs/fluidpatcher #88: error on install script (pip install )

bibitte, 2023-11-01:
On latest Raspbery pi OS 64bit lite. I have an error on install script.

The install script output :

```
$ curl -L git.io/squishbox | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 14510  100 14510    0     0  18238      0 --:--:-- --:--:-- --:--:-- 18238

            o
     o───┐  │  o
      ___│__│__│___
     /             \  o   SquishBox/Headless Pi Synth Installer
 o───┤  _________  │  │     by GEEK FUNK LABS
     │ │ █ │ █ █ │ ├──┘     geekfunklabs.com
     │ │ █ │ █ █ │ │
     \_│_│_│_│_│_│_/

This script installs/updates software and optional extras
for the SquishBox or headless Raspberry Pi synth.
Always be careful when running scripts and commands copied
from the internet. Ensure they are from a trusted source.
If you want to see what this script does before running it,
hit ctrl-C and enter 'curl -L git.io/squishbox | more'
View the full source code at
https://github.com/GeekFunkLabs/fluidpatcher
Report issues with this script at
https://github.com/GeekFunkLabs/fluidpatcher/issues

Choose your install options. Empty responses will use the [default options].
Setup will begin after all questions are answered.

This software is designed for the latest Raspberry Pi OS (Bullseye),
which does not appear to be the situation here. YMMV!
Proceed anyway? (y/[n]) y
What are you setting up?
  1. SquishBox
  2. Headless Raspberry Pi Synth
Choose [1] 1
What version of SquishBox hardware are you using?
  v6 - Green PCB with SMT components
  v4 - Purple PCB, has 2 resistors and LED
  v3 - Purple PCB, has 1 resistor
  v2 - Hackaday/perfboard build
Choose [v6] 
Enter install location [/home/xxxxxxx] 
Install/update synthesizer software? ([y]/n) 
Update/upgrade your operating system? ([y]/n) 
Which audio output would you like to use?
  0. No change
  1. Default
  2. Headphones
  3. sndrpihifiberry
  4. vc4hdmi0
  5. vc4hdmi1
Choose [3] 
Set up web-based file manager? ([y]/n) 
  Please create a user name and password.
    username: xxxxxxx
    password: xxxxxxxxxxxxxxx
Download and install ~400MB of additional soundfonts? ([y]/n) 

Option selection complete. Proceed with installation? ([y]/n) 

This may take some time ... go make some coffee.

Installing/Updating supporting software...
Updating package indexes...
Upgrading your system...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Installing package python3-pip...
Installing package fluid-soundfont-gm...
Installing package ladspa-sdk...
Installing package swh-plugins...
Installing package tap-plugins...
Installing package wah-plugins...
Installing Python module oyaml...
Failed to install oyaml!
```

if I try to run pip install command I have si error :
```
$ pip install "oyaml"
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

```

I have fixed by removing the file /usr/lib/python3.11/EXTERNALLY-MANAGED : 
```
$ sudo rm /usr/lib/python3.11/EXTERNALLY-MANAGED
```
And now the install works fine.

albedozero, 2023-11-01:
This fix is probably fine, but I'm working on a patch that will do things the "right" way. More discussion in #86

albedozero, 2023-11-18:
Fixed as of [bfc11f0](https://github.com/GeekFunkLabs/fluidpatcher/commit/bfc11f0ccfe1293f18fc3fa583105f5c02be75d3)

closed 2023-11-18 by albedozero