GeekFunkLabs/fluidpatcher #111: No Audio from headlesspi.py on Raspberry Pi

ztaylor4, 2024-01-27:
**Describe the bug**
I'm running on a Raspberry Pi 3B with a Keystep 37.  Notes can make it into the Pi, and notes can make it out separately from midi files, but I can't get the notes playing to produce any sound.

Notes show up with `aseqdump`, e.g.:
```
pi@keystep:~ $ aseqdump -p 24
Cannot connect from port 24:0 - Invalid argument
pi@keystep:~ $ aseqdump -p 20
Waiting for data. Press Ctrl+C to end.
Source  Event                  Ch  Data
 20:0   Note on                 7, note 71, velocity 80
 20:0   Note off                7, note 71, velocity 64
``` 

And if I run a command like the following, I hear audio from the connected headphone jack:

    fluidsynth -a alsa -o audio.alsa.device=plughw:0,0 -n -i /usr/share/sounds/sf2/FluidR3_GM.sf2 SquishBox/midi/elevatorgroove.mid

Because I had to add that `-o` flag, I have edited my fluidsettings to include `audio.alsa.device: 'plughw:0,0'`:

```
soundfontdir: SquishBox/sf2
bankdir: SquishBox/banks
mfilesdir: SquishBox/midi
plugindir: /usr/lib/ladspa
currentbank: bank1.yaml
fluidsettings:
  audio.driver: alsa
  audio.alsa.device: 'plughw:0,0'
  audio.period-size: 64
  audio.periods: 3
  midi.autoconnect: 1
  player.reset-synth: 0
  synth.audio-groups: 16
  synth.cpu-cores: 4
  synth.ladspa.active: 1
  synth.polyphony: 128
```

If I add a few print statements, I can see the `listener` from `headlesspi.py` firing for notes, but I don't see `send_event` functions elsewhere getting called.

Any ideas what I could try next?

**Background Info**
- I am trying to run the program via `headlesspi.py`.
- Running on Raspberry Pi 3B with Debian GNU/Linux 12 (bookworm)
- FluidSynth version 2.3.1

albedozero, 2024-01-27:
Thanks for the helpful info - it appears from your `aseqdump` output that your Keystep is sending notes on MIDI channel 7. The _bank1.yaml_ bank file assumes your keyboard is sending on channel 1. You can change this on your keystep or add the following rule to the `router_rules` section at the top of the bank file:

```yaml
  - {type: note, chan: 7=1}
```

Side note - if you're using headlesspi.py, you may have installed the [legacy version](https://github.com/albedozero/fluidpatcher) of the RPi synth software. I'm trying to disentangle fluidpatcher from the [squishbox](https://github.com/GeekFunkLabs) software - apologies for any confusion.

closed 2024-01-27 by albedozero

ztaylor4, 2024-01-27:
Thank you for finding that so quickly!  I tried a lot of things but hadn't found that setting in *bank1.yaml*.

In general, I was trying to go off of the steps from [Headless Raspberry Pi Synthesizer Update video](https://www.youtube.com/watch?v=ONuXCzcuj7E), but maybe I installed something from the other project when I was struggling and didn't realize how all the parts fit together.  The YouTube videos have been very helpful--thanks for all the work you've put into this project, both developing it and sharing it!